gwlim/openwrt-sfe-flowoffload-ath79 #123: Archer C7 v4 throughput higher on DD-WRT SFE than on OpenWRT SFE

04dco, 2020-04-18:
DD-WRT also has SFE and it is enabled by default. I get higher WAN throughput with their implementation.
https://dd-wrt.com/support/other-downloads/?path=betas%2F2020%2F04-18-2020-r42925%2Ftplink_archer-c7-v4%2F

DD-WRT: https://www.speedtest.net/result/8021161826.png
OpenWRT: https://www.speedtest.net/result/9078438403.png

gwlim, 2020-04-18:
1) Do not use speedtest, the most accurate test is if you run both client and server yourself
2) If you must use a web test use https://fast.com/ instead, it is both IPv4 and IPv6 capable and their servers are better maintained

04dco, 2020-04-18:
OpenWRT: [image]
https://www.dslreports.com/speedtest/62441297

DD-WRT: [image]
https://www.dslreports.com/speedtest/62442055

All tests were done with the default configuration and WiFi disabled.

gwlim, 2020-05-07:
Added new option
echo 1 > /sys/fast_classifier/skip_ingress_qdisc

closed 2020-05-07 by gwlim

MrObvious, 2020-05-14:
Is this on the May build?

gwlim, 2020-05-14:
Yes although I still don't think the issue is with SFE
When the router just booted a lot processes are eating up the cpu, you should let the router idle and the line idle for a while before testing