gwlim/openwrt-sfe-flowoffload-ath79 #129: Build from source

black-cerberon, 2020-05-08:
Hi. Can I build it form source itself? I don't need SQM, bandwidth monitor, USB support (router w/o USB), etc.

dag03tsc, 2020-07-08:
Hi @black-cerberon !

It's been asked before as in #9 #57 or #64 . I actually supported this in #9 but, as @gwlim himself told: 

> Last time I did that didn't turn out well, I guess people are impossible to please.
You can still take the patches off the old repo.

Anyway, I guess it's worth the try. Let's hope he changed his mind about this.

I asked for this because, with my WDR-4900v1 (mpc85xx), and due to my ISP configuration, I need to apply some additional custom patches and packages to get all services working. Not applying them would mean to partially or fully lose VoIP and IPTV hired services, and that's a no go.

A couple of years ago I managed to get it running perfect and smoothly using v17.01.4 and old patches but, of course, that's an obsolete version with no support (no more updates or patches as new functionalities, bug and security fixes, performance improvements, etc.). So, it isn't actually a long-term solution but hey, it's been working flawlessly with that image as my main router since then!

There's already a tar file containing all required kmod packages so the only thing missing and required to replicate a build would be the source of all patches.

@gwlim , I can see your point of view but we're not asking you to allow pull requests or things like that. We just want to be able to build our own images based on some or all of your updated patches and packages. A simple tar containing all patch folders would be awesome, as you already do with the packages folder. We really hope you change your mind!

Anyway, thanks a lot for your hard work! Thought I haven't been able to use your updated builds due to my needs, which is a shame. Please, please, @gwlim , publish patches sources!!

I offer you my help if you think I could make anything to keep this maintained; be welcome to ask, and I'm pretty sure there are more people like me that would be glad to help to maintain this if you agree.

Once again, thanks a lot for your hard work to keep all this maintained and evolving!

black-cerberon, 2020-07-08:
@dag03tsc,in first, this binary builds may contain some malware or backdoor, whatever. I don't trust for it, before he not share the sources. In second, this builds contains many unwanted packages for me. 

About GPL, he must share sources of used software with GPL licence, at least name, version, etc. But he close all code, not only shortcut-fe with fast-classifier module. Also, this guy has bookmarks of NSA repos. This is a looking very-very weird.

supersn0b, 2020-07-12:
Yes, this does really not make any sense. I also wanted to start building this on my own now and removing and adding some packages and maybe tweak some configs as I have some devices that cannot connect via wifi. @gwlim please publish the sources! In the meanwhile, I'm going back to good old OpenWrt...

dag03tsc, 2020-08-15:
> @dag03tsc,in first, this binary builds may contain some malware or backdoor, whatever. I don't trust for it, before he not share the sources. In second, this builds contains many unwanted packages [...]

@black-cerberon, your right, and that's one of the reasons I rather build and patch my own images, so I can pre-review what the patches actually change, apply only the ones that I need (as I do with packages too) and add my own ones.
That's the reason I've never flashed any of the builded images.

I actually took the patches from the old repo as @gwlim suggested and I used the v17.01.4 branch.
I modified a few so they could be applied to newer versions of files and deleted a couple of old broken ones.
Then I applied my custom patches and diffconfig file and everything has been smooth so far with that build as my main router (for about 3 years or so if I recall it right).

Without SFE and Fast-Path, the I/O speed through WAN NAT dropped to about ~320Mb/s due to the HWNAT feature not being supported for this platform (mpc85xx). With both SFE and Fast-Path enabled and tuned right, I can fully use the 600Mb/s that my ISP provides (about ~630Mb/s in fact).

mrfixit2001, 2020-12-04:
Where can I find the patches? Not seeing the "old repo" with them in it.

fkleon, 2020-12-06:
I'd also be keen to build this from source to get an up-to-date release.

@gwlim Licence issues aside, it would be great if you could publish the patches so that this popular fork doesn't solely rely on you for fresh releases (good old bus factor!). I'm sure the community would appreciate it.

black-cerberon, 2020-12-06:
You can take fast-classifier and shortcut-fe with patches from [https://github.com/coolsnowwolf/lede](this) repo and build with clean mainsteam Openwrt. Don't forget make init.d script for shortcut-fe.

Priyantha, 2021-05-19:
> You can take fast-classifier and shortcut-fe with patches from [this](https://github.com/coolsnowwolf/lede) repo and build with clean mainsteam Openwrt. Don't forget make init.d script for shortcut-fe.

This is a good suggestion :-)

But I am unable to get it build properly :(
Did you test it recently? for example against the current OpenWRT 21.02 tag?
It looks like the kernel patch for shortcut-fe doesn't work.