gwlim/openwrt-sfe-flowoffload-ath79 #130: 2020 March and May builds on C7

eisengrau, 2020-05-09:
Hello,

It looks like something is not OK with your March and May builds. I have a C7v5, my friend has a v2, he also reported the same issues:

- coming from sysupgrade from the FEB build, the MAR image has stability issues with 5Ghz, frequent disconnects
- also I tried to flash the new MAY build via TFTP, there is link on LAN, but it seems like ICMP ping does not reply, there is no SSH, no LuCI... . Tried also by retruning back to stock TP-link firmware -> your build. Same issue. I can only return to OpenWRT now by going back to stock -> some old build of Openwrt (ar71xx branch) -> sysupgrade via CLI.

So far, the best results we had were with your 2020 FEB build on these devices.

gwlim, 2020-05-10:
Have you tried not saving the configs?

eisengrau, 2020-05-11:
Seems to be fine now, however I was only able to return to tp-link factory -> upgrade to 18.x.x vanilla -> 19.x.x ath79 -> sfe-offload MAY. Upgrading directly from FEB it seems that it trashed the system.

I always upgrade the following way: backup settings -> flash sysupgrade image - not keeping settings -> do another factory reset and reboot -> enable/disable wanted services, reboot -> import settings, or start fresh and reboot.

Your tip on importing the settings makes sense, I'll definately try fresh from now on.

closed 2020-05-11 by gwlim