gwlim/openwrt-sfe-flowoffload-ath79 #131: Cannot deploy SQM, missing/incompatible packages

KingRabbid, 2020-05-09:
Hello!

I've used https://github.com/gwlim/openwrt-sfe-flowoffload/tree/master/MAY-2020/openwrt-sfe-flowoffload-mini/TP-Link%20WR1043NDv1-32MB-RAM-mips24k-ath9k and now I have:

> Model	TP-Link TL-WR1043ND v1
> Architecture	Atheros AR9132 rev 2
> Firmware Version	OpenWrt SFE r13172-2b31f143f9 / LuCI Trunk
> Kernel Version	5.4.39

I'm trying to install SQM but constantly failing for both _luci-app-sqm_ and _sqm-scripts_. Am I doing something wrong or SQM is not supported anymore for this router?

Thank you

[image]
[image]
[image]
[image]

gwlim, 2020-05-10:
You need to install all the required kmods from the tar.gz file first

KingRabbid, 2020-05-10:
Hi,

thanks for quick reply; I've managed to identify the following in the _kmod-package.tar.gz_ available in git:
```
root@OpenWrt:/tmp# ll *.ipk
-r-xr-x---    1 root     root         13514 May 10 23:04 iptables-mod-ipopt_1.8.4-1_mips_24kc.ipk*
-r-xr-x---    1 root     root         14354 May 10 23:09 kmod-ipt-conntrack-extra_5.4.39-1_mips_24kc.ipk*
-r-xr-x---    1 root     root          6822 May 10 23:05 kmod-ipt-ipopt_5.4.39-1_mips_24kc.ipk*
-r-xr-x---    1 root     root          2393 May 10 23:06 kmod-ipt-raw6_5.4.39-1_mips_24kc.ipk*
-r-xr-x---    1 root     root          2359 May 10 23:06 kmod-ipt-raw_5.4.39-1_mips_24kc.ipk*
-r-xr-x---    1 root     root         10210 May 10 23:23 kmod-lib-zlib-deflate_5.4.39-1_mips_24kc.ipk*
-r-xr-x---    1 root     root          8415 May 10 23:23 kmod-lib-zlib-inflate_5.4.39-1_mips_24kc.ipk*
-r-xr-x---    1 root     root         16159 May 10 23:07 kmod-sched-cake_5.4.39-1_mips_24kc.ipk*
```
How do I know which package is in this archive contains the remaining required dependencies:
> tc
> iptables-mod-conntrack-extra
> libelf1
> libmnl0
Or I can download the latest snapshot version for these?

gwlim, 2020-05-11:
non kmod packages should be installable from latest snapshot

gwlim, 2020-05-11:
Anyway I suggest upgrading the RAM by replacing the DDR chip
https://openwrt.org/toh/tp-link/tl-wr1043nd#mb_ram_mod

KingRabbid, 2020-05-11:
Many thanks, installation worked based on opkg update + install of the mentioned packages.
Now I can see SQM web page(s).
I'll look into the RAM thing if it's something I can handle.

Thanks again and regards!

closed 2020-05-11 by KingRabbid