gwlim/openwrt-sfe-flowoffload-ath79 #133: WNDRMACv2 missing

cybik, 2020-05-10:
The Netgear WNDRMACv2 is a router based off the WNDRMAC3800 hardware, as per https://openwrt.org/toh/hwdata/netgear/netgear_wndrmac_v2 . Oddly enough, I am seeing the WNDR3800 on the snapshots list for ath79, but the WNDRMACv2 is nowhere to be found.

I can likely just flash the 3800 and be done with it, but i'd prefer it if I could keep the software and hardware branding aligned.

cybik, 2020-05-10:
If it's any help, I just opened https://github.com/openwrt/openwrt/pull/3010 to possibly introduce WNDRMAC* in ath79. As stated, fully untested.

gwlim, 2020-05-11:
Right, will build when support is committed

cybik, 2020-06-11:
Support for v2 [and even v1] is in Upstream now :D

https://github.com/openwrt/openwrt/commit/7f297e740bc288d29fab1f7490ed0e2e49e9e6f9
https://github.com/openwrt/openwrt/commit/108df3eabbb4dd93ff3c616c9455d69daaa98b49

gwlim, 2020-06-24:
Added in July Image

closed 2020-06-24 by gwlim