gwlim/openwrt-sfe-flowoffload-ath79 #137: usign missing

cybik, 2020-06-24:
When running `opkg update` from command line, openwrt complains with `usign: not found`.

You might want to pull in https://openwrt.org/packages/pkgdata/usign when building the bins.

cybik, 2020-06-24:
The fix is to manually obtain the package from https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/base/ , install it using console, then opkg update. This looks like it works, a bit.

Zsolteeca, 2020-06-24:
So in Luci can't you install any package?

cybik, 2020-06-24:
Not until I installed the package manually.

Zsolteeca, 2020-06-24:
But I would like to install luci-app images and I got the code 255 error code
For example.. "pkg_init_from_file: Failed to extract control file from https://downloads.openwrt.org/snapshots/packages/mips_24kc/luci/luci-app-simple-adblock_git-20.174.75173-8ff50bf_all.ipk."

You? Can you install these packages to Luci?

cybik, 2020-06-24:
After I manually downloaded usign using the url https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/base/usign_2020-05-23-f1f65026-1_mipsel_24kc.ipk , then scp'd it to my device and installed it, I could run opkg update and download other packages again.

Zsolteeca, 2020-06-24:
Strange..
I had installed the latest version of this firmware to my Linksys WRT1043ND v2 device and I can install any package without the package named "usign"..
What device do you use?

cybik, 2020-06-24:
WNDRMACv2. It's one of the newest additions to the mix.

I also have a quirky configuration and needed to add a bunch of packages before uhttpd even showed up XD

Zsolteeca, 2020-06-24:
Sorry I have just installed some packages from luci-app branch.. So I don't know what kind of components does the base section require...

gwlim, 2020-06-24:
It is to save space by not installing usign

cybik, 2020-06-25:
Indeed I have.

closed 2020-06-25 by gwlim