gwlim/openwrt-sfe-flowoffload-ath79 #139: Feedback On Achieved NAT/Routing Performance

gwlim, 2020-06-28:
I would like to know the NAT/Routing performance of hardware achieved
1) AR9344 can hit 1Gbps @ 1500MTU (Overclocked to 720 MHZ)
2) QCA9563 can also hit 1Gbps @ 1500MTU
3) AR9132 can hit 700Mbps @ 1500MTU (Overclocked to 430MHZ)
Any other data especially AR7161 CPUs?

eisengrau, 2020-07-01:
Hello,

Is it possible to run an iperf test on the device itself? Or that would be unrealistic? I assume the best way is the classic: 1 client on the LAN side, one on the WAN side, both running iperf.

gwlim, 2020-07-01:
If you have 1Gbps WAN ISP subscription https://fast.com/ would do
If not 
https://openwrt.org/docs/guide-user/perf_and_log/benchmark.nat
https://openwrt.org/docs/guide-user/perf_and_log/benchmark.ipv6.routing

MrObvious, 2020-07-20:
With SFE, I can hit 900-950 down using Ookla on my Archer C7 V2.

gwlim, 2020-07-23:
> With SFE, I can hit 900-950 down using Ookla on my Archer C7 V2.

Thanks, I have the data on mips74kc routers as I owned a couple of mips74k routers. I am more curious about mips24k routers especially the AR7161 routers

eisengrau, 2020-07-27:
Hallo,

> Model
> MikroTik RouterBOARD 450G
> Processor
> Atheros AR7161 rev 2
> Architecture
> MIPS 24Kc V7.4
> Firmware Version
> OpenWrt SFE r10912-d94a3f328f / LuCI ( Trunk ) 

Please see #20 

I am still using your last available on this device, and it is still my 'daily driver' at the moment.

closed 2021-01-22 by gwlim