gwlim/openwrt-sfe-flowoffload-ath79 #143: Xiaomi R3G support

kolomparrudi, 2020-08-01:
Hi gwlim

Your Openwrt SFE image is running in my 1043ngv2 router with 246 days uptime.

I have MIR3G router to receive, but there is no good openwrt alternative for it. Built in flow offload is buggy (port forward issues), without it the throughput is lower than my old router.

It would be also fine if you can publish your sfe related patches and I'll try to port it.

Thanks,
kolomparrudi

eisengrau, 2020-08-03:
AFAIK shortcut-fe and offload are features only for Atheros-based devices. As the R3G and all of Xiaomi's other home network devices are Mediatek based.

I also have an AC2100 (white variant with antennaes) and have OpenWrt running on it (made by a user called scp07 - you can head over to the openwrt forums to check it out). I have been running this since months and it's stable since I know it. Of course I'm using it as a network bridge or 'dumb AP', so no routing/firewall.

kolomparrudi, 2020-08-03:
No, shortcut-fe is a device and architecture independent forwarding engine implementation.
I have already started to port it based on old published lede patches and dd-wrt implementations, but it would be easier to save to issues what Gwlim already solved.

eisengrau, 2020-08-03:
Oh, cool, I didn't knew that! In that case, I'm eager to test that out, once to get up to speed wit it. Do you have a repo with the code already?

kolomparrudi, 2020-08-03:
Currently I have it on my server. Trying to compile and publish it when I have a working version

gwlim, 2020-08-05:
I am working on mt7621 once I have sorted out the stability issues and feel it is ready I will release create for mt7621

kolomparrudi, 2020-08-05:
I have already ported sfe to openwrt trunk, I have solved all compile issues, now I'll start testing it. Created an openwrt fort and pushed my changes to it:
https://github.com/kolomparrudi/openwrt

gwlim, 2021-01-22:
Added in latest release

closed 2021-01-22 by gwlim