gwlim/openwrt-sfe-flowoffload-ath79 #145: nlbwmon shows values almost 4 times higher than they should be

macearl, 2020-08-19:
i recently enabled nlbwmon and noticed after a few days that the values were extremely high.
So after a database reset i just downloaded a 10GB file from one client to see what nlbwmon would report

```
wget -O /dev/null http://speedtest.tele2.net/10GB.zip
```
afterwards nlbwmon reports `38.44 GB` of downloaded data.

I'm running the July 2020 image with default nlbwmon settings on an Archer C7 v2

jkhsjdhjs, 2020-08-19:
Can confirm this issue. nlbwmon reports more than 56GB after downloading the speedtest file specified above. Upload values are also way too high. Also running July 2020 image with default nlbwmon settings on an Archer C7 v5.

zigagrcar, 2020-09-27:
I had the same problem with previus versions too, I suggest using VnStat Traffic Monitor it does not give per device reports but its okay for overall monthly usage.

gwlim, 2021-01-22:
Fixed in latest release

closed 2021-01-22 by gwlim