gwlim/openwrt-sfe-flowoffload-ath79 #148: wdr4900 buid

cosminsstoica, 2020-11-13:
Hi,
I'm using the latest build (JUL-2020) for Tp-Link WDR4900v1 on a 1000/500 Mbps fiber internet connection. 
Any ideea why Windows clients have lower Download/Upload speed than Linux clients? 
The minimum configuration for the PC is:  I5-8300/i7-8700, 8/16GB RAM and 256/512 nvme.
Not using SQM QOS, and SFE flow offloading is enabled.

[image]

gwlim, 2020-11-19:
A few things you can check
1. Congestion Control Inside netsh
https://techcommunity.microsoft.com/t5/networking-blog/core-network-stack-features-in-the-creators-update-for-windows/ba-p/339676
2. Windows Device Driver
I have experience before certain realtek driver doesn't do gigabit data transfer but others do

cosminsstoica, 2020-12-02:
After days of tuning I found "SG TCP Optimizer", which helped me a lot. Still don't get as much speed as on Linux, but now I can reach 800 Mbps upload / 450 Mbps download on Windows. 
I thought it was time to change the router, but I'll keep it for a while. 

Tnx

closed 2020-12-02 by cosminsstoica