gwlim/openwrt-sfe-flowoffload-ath79 #151: 2021-01 TP-Link Archer C7 v5 firmware has incompatible kernel modules

jkhsjdhjs, 2021-01-23:
It seems that the kernel modules of the new release packaged for the TP Link Archer C7 v5 are incompatible with the firmware.

I just upgraded my router's firmware to the January 2021 release, using the sysupgrade file from this directory: https://github.com/gwlim/openwrt-sfe-flowoffload-ath79/tree/master/2021-01/openwrt-sfe-flowoffload-normal/mips74k/TP-Link%20Archer%20C7v5-mips74k-ath10k

Then I tried to install the `kmod-iptunnel6` kernel module from the `kmod-package.tar.gz` archive, which I was unable to do because of the following error:
```
root@OpenWrt:/tmp# opkg install kmod-iptunnel6_5.4.91-1_mips_24kc.ipk 
Unknown package 'kmod-iptunnel6'.
Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-iptunnel6 found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-iptunnel6.
```

The file is actually located in the `/tmp` directory, so this is not an issue. It also worked fine for previously releases.

gwlim, 2021-01-23:
It should be
`opkg install /tmp/kmod-iptunnel6_5.4.91-1_mips_24kc.ipk`

jkhsjdhjs, 2021-01-23:
Thanks for your reply, this was actually the first way I tried it. I only changed directory to `/tmp`, after it didn't work. So unfortunately this doesn't work either.
I just downgraded to the July 2020, there it also worked by changing to the `/tmp` directory and specifing the filenames in `opkg install` without a directory.

gwlim, 2021-01-23:
Did you use tab to autocomplete?
I verified the packages and checksum pretty sure I didn't copy wrongly.

jkhsjdhjs, 2021-01-23:
Yes, I used tab completion. But I'll try upgrading and installing the kernel modules again tomorrow, just to be sure I didn't mess up somewhere. I will report back here.

jkhsjdhjs, 2021-01-23:
I don't know what the hell I did wrong yesterday, but it worked now. Thanks and sorry for the noise.

closed 2021-01-23 by jkhsjdhjs