gwlim/openwrt-sfe-flowoffload-ath79 #153: mt7621 flash issue

eisengrau, 2021-01-24:
Hello,

Thank you for your great effort in providing new images for the new year!

I noticed you have released an mt7621 compatible sysupgrade image as well. I tried to flash it on my Redmi AC2100 (white with external antennaes), however I get the below error when I do a sysupgrade -v -n:

>  Device xiaomi,redmi-router-ac2100 not supported by this image
> Supported devices: xiaomi,redmi-router-ac2100 - Image version mismatch: image 1.1, device 1.0. Please wipe config during upgrade (force required) or reinstall. Reason: Config cannot be migrated from swconfig to DSA
> Image check failed.

I'm not sure about the version 1.1 hardware, if this is reason of the error, as far as I am aware, you can get the same device at the moment from the chinenese marketplaces (I have 2 of these currently).

I'm running the below image on them currently:

[image]

gwlim, 2021-01-24:
I don't have the device so I am not sure about the difference in hardware version.
Is it recoverable? If so you can try.
I am basing my testing on MT7621 platform on Newifi D2 and it has BREED installed so it is always recoverable

eisengrau, 2021-01-24:
Oh yes, I have BREED on them as well. I was just hesitant to force flash it due to the error. Perhaps I can try any way, nothing to lose here. However if it's indeed the case that DSA is now the standard in this image, I may postpone it, as I use VLANs, and as I understood that there is no easy way at the moment to configure VLAN tagging with DSA (LuCI does not have a "Switch" config page either).