gwlim/openwrt-sfe-flowoffload-ath79 #154: Xiaomi R3G V1 - no module folders for kernel version 5.4.52 found

kolomparrudi, 2021-01-25:
Hi Gwlim,

Upgraded to the latest build on my Xiaomi R3G v1 router, seems to be kernel version mismatch:

logread output:
Mon Jan 25 13:16:40 2021 daemon.err modprobe: no module folders for kernel version 5.4.52 found
Mon Jan 25 13:16:40 2021 daemon.err modprobe: no module folders for kernel version 5.4.52 found
Mon Jan 25 13:16:40 2021 daemon.err modprobe: no module folders for kernel version 5.4.52 found
Mon Jan 25 13:16:40 2021 daemon.err modprobe: no module folders for kernel version 5.4.52 found

dmesg output:
[   11.874703] procd: - early -
[   11.877726] procd: - watchdog -
[   12.437475] procd: - watchdog -
[   12.441066] procd: - ubus -
[   12.501837] procd: - init -
[   13.002466] kmodloader: no module folders for kernel version 5.4.52 found

root@lili:~# ls -l /lib/modules/
drwxr-xr-x    2 root     root          3006 Jan 21 22:56 **5.4.91**

root@lili:~# lsmod
root@lili:~#

MrObvious, 2021-01-25:
Use the kmod tar file for what you need. It won't be in opkg.

kolomparrudi, 2021-01-25:
Not installed anything, this is the original state after flashing.
Image shipped with kernel 5.4.52, but kernel modules are 5.4.91:
root@lili:~# uname -a
Linux lili **5.4.52** #0 SMP Sun Aug 9 22:01:51 2020 mips GNU/Linux

root@lili:~# ls -l /lib/modules/
drwxr-xr-x    2 root     root          3006 Jan 21 22:56 **5.4.91**

kolomparrudi, 2021-01-25:
I think it's not the image problem. It seems the kernel remained the old, what was used previously. Maybe device booted from different partition. Trying to investigate it.

kolomparrudi, 2021-01-25:
Tried to set autoboot.command to 0x600000 or to 0x200000, but it won't boot it these value was set.
Can you provide kernel0 and rootfs1 images to be able to create breed recovery image?

MrObvious, 2021-01-26:
Which did you install? uname should return 5.4.91. For example on my Archer C7V2:
root@OpenWrt:~# uname -a
Linux OpenWrt 5.4.91 #0 Fri Jan 22 10:36:49 2021 mips GNU/Linux
root@OpenWrt:~#

kolomparrudi, 2021-01-26:
Insalled the latest 2021 version.  I know, it should have 5.4.91.
My router booted from /dev/mtd7 (kernel_stock), but sysupgrade installed kernel to /dev/mtd8 (kernel).
Previously I have installed openwrt trunk from Breed, maybe some nvram settings cause the problem.

kolomparrudi, 2021-01-26:
Figured out, I have to set this environment variable in Breed to boot from /dev/mtd8:
autoboot.command "boot flash 0x600000"

Unfortunately when the router booted with the right kernel, it never goes up.
Unfortunately I don't have serial console, so I can't tell you more details.
I have tested it with the latest openwrt snapshot and it's working. When I upgrade to this SFE build, it won't boot anymore.

gwlim, 2021-01-27:
Any logs for this?

kolomparrudi, 2021-01-27:
Without serial consdon't have any logs :(
When sysupgrade started, the upgrade seems to finish properly, router rebooted, seen the bootloader for some seconds (replied to some ping pockets from 192.168.1.1), but no life signals after this.

If you can provide test builds, I can test it.

Thanks,

eisengrau, 2021-01-30:
I was also curious how did you manage to flash this image? Did you do sysupgrade --force *.bin? As it has a different partition layout that the tp-link devices, the image format should be something, as you mentioned, with kernel1 and rootfs0 type, same format as BREED also accepts...

Nevertheless, I did probably what you did too, and got my Redmi AC2100 unbootable right now. The issue could be that if we could make the images partition compatible.

> 
> 
> Without serial consdon't have any logs :(
> When sysupgrade started, the upgrade seems to finish properly, router rebooted, seen the bootloader for some seconds (replied to some ping pockets from 192.168.1.1), but no life signals after this.
> 
> If you can provide test builds, I can test it.
> 
> Thanks,

kolomparrudi, 2021-01-30:
> I was also curious how did you manage to flash this image? Did you do sysupgrade --force *.bin? As it has a different partition layout that the tp-link devices, the image format should be something, [...]

I have a create_openwrt_for_breed.cmd found on R3G telegram channel, it create breed image from kernel1 and rootfs0 in the following way:
1. Extend kernel1 to 4194304 size by adding zeros (this is the mtd partition size)
2. Concat the image from the first step twice
3. Append rootfs0 to them
Here is the source:
```
@echo off
setlocal enableextensions enabledelayedexpansion

set "dirpath_current=%~dp0\"
set "pathname_kernel=%dirpath_current%openwrt-ramips-mt7621-xiaomi_mir3g-squashfs-kernel1.bin"
set "pathname_rootfs=%dirpath_current%openwrt-ramips-mt7621-xiaomi_mir3g-squashfs-rootfs0.bin"
set "pathname_breed=%dirpath_current%openwrt-ramips-mt7621-xiaomi_mir3g-breed.bin"
if "%~3" neq "" set "pathname_breed=%~3"
if "%~2" neq "" set "pathname_rootfs=%~2"
if "%~1" neq "" set "pathname_kernel=%~1"
if not exist "%pathname_rootfs%" echo File "%pathname_rootfs%" not found & goto help
if not exist "%pathname_kernel%" echo File "%pathname_kernel%" not found & goto help
set "pathname_temp=%TEMP%\~tempfile"
set "pathname_ktmp=%TEMP%\~kernel"
del /q /f "%pathname_breed%" >nul 2>nul
del /q /f "%pathname_temp%" >nul 2>nul
del /q /f "%pathname_ktmp%*" >nul 2>nul
set kernel_size=0
for %%I in ("%pathname_kernel%") do set/A kernel_size=4194304-%%~zI
fsutil file createnew "%pathname_temp%" %kernel_size%
copy /b "%pathname_kernel%"+"%pathname_temp%" "%pathname_ktmp%"
del /q /f "%pathname_temp%" >nul
copy /b "%pathname_ktmp%"+"%pathname_ktmp%" "%pathname_ktmp%1"
copy /b "%pathname_ktmp%1"+"%pathname_rootfs%" "%pathname_breed%"
del /q /f "%pathname_ktmp%*" >nul

goto :endofscript

:help
echo.
echo.This will make OpenWrt binary file for using with Breed for Mi Router 3G.
echo.
echo Usage:
echo create_openwrt_for_breed.cmd [kernel1 file] [rootfs0 file] [breed file]
echo.
echo Example: %0
echo create_openwrt_for_breed.cmd openwrt-18.06.0-ramips-mt7621-mir3g-squashfs-kernel1.bin openwrt-18.06.0-ramips-mt7621-mir3g-squashfs-rootfs0.bin
echo.
pause

:endofscript
endlocal
```
I used this method to install openwrt trunk to my router and I tried to upgrade from this to SFE build. I don't have to use the force option with sysupgrade.

tmuehlhoff, 2023-05-10:
I have a similar problem. When Updating via LuCI the kernel seems to be loaded from the wrong partition, how can I fix that?! Thanks.

MrObvious, 2023-05-10:
I gave up and went back to just regular OpenWRT. This hasn't been updated since 2021 and there are security vulnerabilities. The regular version works well enough now that I don't notice much of any difference.