gwlim/openwrt-sfe-flowoffload-ath79 #167: Add support for EdgeRouter X

delet-this, 2021-07-13:
Could you add support for Ubiquiti EdgeRouter X? It's a MT7621 device

https://openwrt.org/toh/hwdata/ubiquiti/ubiquiti_edgerouter_x

OUZYcn, 2021-10-31:
It look like you have a MT7621 (mips1004k) build, can you enable below hardware also?

Target hardware：EdgeRouter X (ER-X)
SoC：MediaTek MT7621AT **mips1004kc**
architecture ：MIPSel

OpenWRT：https://openwrt.org/toh/ubiquiti/edgerouter_x_er-x_ka

OUZYcn, 2021-11-01:
> It look like you have a MT7621 (mips1004k) build, can you enable below hardware also?
> 
> Target hardware：EdgeRouter X (ER-X) SoC：MediaTek MT7621AT **mips1004kc** architecture ：MIPSel
> 
> OpenWRT：https://openwrt.org/toh/ubiquiti/edgerouter_x_er-x_ka

Complied by myself today but can’t put sfe and flow offloading in same firmware. One option a time.